FIU-SEDS/Vensors #7: Write an error sources technical report for time of flight sensors

luis-c465, 2025-03-04:
The Time of flight sensor (VL53L1X) is going to be used to measure the distance of the carriage at every point in time. The second derivative of these discrete points is going to be calculated.

Given that the sensor will be set in short range mode, when the second derivative is calculated what is the maximum expected deviation of the acceleration from the expected value?

For e.x: +- 0.05 m/s^2.

I.e. what is the accuracy of the sensor in finding the acceleration of a moving object given that the sensor has a specific accuracy in the datasheet bellow. 

You will write a document about your findings, including the maximum expected acceleration deviation, and the math behind your findings. 

Additionally, should you find ways to improve accuracy when writing the code and or installing the sensors, please document them as well in the report.

See the datasheet here: https://www.st.com/resource/en/datasheet/vl53l1x.pdf

Amazon link: https://www.amazon.com/dp/B09V4DS888

Kevin-Parra271, 2025-03-17:
[Technical Report.docx](https://github.com/user-attachments/files/19279832/Technical.Report.docx)

Kevin-Parra271, 2025-03-17:
I used the standard formula of acceleration, where the acceleration is equal to the change of velocity over the change in time (Δv/Δt). I converted the velocity to the chance of distance over time (Δd/Δt), when simplifies the acceleration to the change of distance over the change of time squared (Δd/Δt^2). The change of distance in this case is the max ranging error that I found in the datasheet and the time interval as the frequency conversion of 50hz, which is also at its max. I hope this answers your question.

luis-c465, 2025-03-17:
Yep sounds good, that is useful to know I did some calculations, and it seems that when increasing the time interval the error in the acceleration can be brought back under control.

As a side note since noise from the sensors is going to be a major part of this project I think I will make a task to find out ways to filter it out!

closed 2025-03-17 by luis-c465