FIU-SEDS/Vensors #10: Find libaries and example code for Time of flight sensor

luis-c465, 2025-03-06:
Create a report with a list of compatible libraries and example code with those libraries for the following sensor:

Time of Flight Sensor (VL53L1X)10mm x 25mm x 4mm

VL53L1X

[Amazon Link](https://www.amazon.com/dp/B09V4DS888?ref=cm_sw_r_cp_ud_dp_G8J6C32HENN1VCQNBQWC&ref_=cm_sw_r_cp_ud_dp_G8J6C32HENN1VCQNBQWC&social_share=cm_sw_r_cp_ud_dp_G8J6C32HENN1VCQNBQWC&_encoding=UTF8&psc=1)

luis-c465, 2025-03-09:
@Shamar12334 How is the report going?

Shamar12334, 2025-03-09:
@luis-c465 I have completed the report.Would like me to merge it to GitHub or present it at our next meeting?

luis-c465, 2025-03-09:
If you have the report already could you send attach it as an attachment to a comment here?

Shamar12334, 2025-03-09:
@luis-c465 okay I’ll send it

Shamar12334, 2025-03-10:
[TOF report.docx](https://github.com/user-attachments/files/19153583/TOF.report.docx)

Shamar12334, 2025-03-10:
@luis-c465

Shamar12334, 2025-03-10:
@luis-c465 updated-https://fiudit-my.sharepoint.com/:w:/g/personal/sweek011_fiu_edu/Eafx337xaxRHqwieH3n97CYB6Zt-PV8M5_i-jQIfzDWOkA?e=jRcakr

closed 2025-03-10 by luis-c465