FIU-SEDS/Vensors #8: Find libaries and example code for Accelerometer

luis-c465, 2025-03-06:
Create a report with a list of compatible libraries and example code with those libraries for the following sensor:

HiLetgo 3pcs GY-521 MPU-6050 MPU6050 3 Axis Accelerometer Gyroscope Module 6 DOF 6-axis Accelerometer Gyroscope Sensor Module 16 Bit AD Converter Data Output IIC I2C for Arduino 3 x 2.5 x 0.5 inches

[Amazon link](https://www.amazon.com/dp/B00LP25V1A?pd_rd_i=B01DK83ZYQ&pd_rd_w=Tmwsv&content-id=amzn1.sym.7446a9d1-25fe-4460-b135-a60336bad2c9&pf_rd_p=7446a9d1-25fe-4460-b135-a60336bad2c9&pf_rd_r=VQ2CYX3MYDKX3GVQ5QES&pd_rd_wg=c9Vsy&pd_rd_r=3eed6e98-d4cd-45f3-8c95-ca6f73fc523c&s=industrial&sp_csd=d2lkZ2V0TmFtZT1zcF9kZXRhaWw&th=1)

Rahinfgit, 2025-03-07:
[Report_ Compatible Libraries and Example Code for HiLetgo GY-521 MPU-6050 Accelerometer & Gyroscope Sensor.pdf](https://github.com/user-attachments/files/19134607/Report_.Compatible.Libraries.and.Example.Code.for.HiLetgo.GY-521.MPU-6050.Accelerometer.Gyroscope.Sensor.pdf) 

Hi luis. This is the report that I created with example code. Let me know if you need further adjustment.

luis-c465, 2025-03-09:
Looks great, I gave it a look through, and we will be going with the AdaFruit library as they have a great reputation and the library is already included in the Arduino IDE.

Should we have any issues with the library taking too much storage space, we can go with Jeff's library.

closed 2025-03-09 by luis-c465