FIU-SEDS/Vensors #9: Find libaries and example code for Accelerometer

luis-c465, 2025-03-06:
Create a report with a list of compatible libraries and example code with those libraries for the following sensor:

Adafruit ADXL345 Triple-Axis Accelerometer (+-2g/4g/8g/16g) w/ I2C/SPI [ADA1231] 25mm x 19mm x 3.14mm

ADXL345

[Amazon Link](https://www.amazon.com/dp/B01BT4N9BC?ref=cm_sw_r_cp_ud_dp_3VSGT5EG49Y75YVY5DWB_1&ie=UTF8&useRedirectOnSuccess=1&ref_=dex_glow_signin&path=/dp/B01BT4N9BC)

Mblen, 2025-03-07:
[Library Names.docx](https://github.com/user-attachments/files/19134609/Library.Names.docx)

Mblen, 2025-03-07:
@luis-c465  I have completed the research. Could you please review it and provide any feedback or suggestions for improvement?"

Let me know if you need further adjustments!

luis-c465, 2025-03-09:
@Mblen 

Looks great, because the sensor is made by Adafruit we will be going with the Adafruit ADXL345 library for better support. 

I just have 2 questions, in the SparkFun ADXL345 sensor you mention it has the tap detection and free-fall detection feature. Can you elaborate a little on that.

Also, could you link below the GitHub to the Adafruit library?

closed 2025-03-13 by luis-c465