FIU-SEDS/Vensors #13: Fill in functions for Time of Flight Sensor

luis-c465, 2025-03-13:
Fill in the functions found in the `timeOfFlight.ino` file for the VL53L1X sensor using the [SparkFun VL53L1x Laser Distance Sensor Library](https://github.com/sparkfun/SparkFun_VL53L1X_Arduino_Library).

The bellow functions should be filled in to return the appropriate values when the functions are called.
Do not worry about which pins the sensors are connected to, that will be handled later.

```ino
/**
 * Returns the distance measured by the left sensor in mm
 */
int getLDistanceSensor()
{
  // Simulate reading from a sensor
  return -1;
}

/**
 * Returns the distance measured by the right sensor in mm
 */
int getRDistanceSensor()
{
  // Simulate reading from a sensor
  return -1;
}

/**
 * Initalized the time of flight sensors and returns a 0 value if successfull.
 * @return 0 on success, non-zero otherwise.
 */
int setupTimeOfFlight()
{
}
```

luis-c465, 2025-03-17:
@Shamar12334 How is the code going?

Shamar12334, 2025-03-17:
@luis-c465  I’m done. I did a pr last night

closed 2025-03-17 by luis-c465