FIU-SEDS/Vensors #15: Write functions for Adafruit accelerometer ADXL345

luis-c465, 2025-03-13:
Fill in the functions found in the `accelerometer.ino` file for the Adafruit ADXL345 sensor using the [Adafruit ADXL345](https://github.com/adafruit/Adafruit_ADXL345).

The bellow functions should be filled in to return the appropriate values when the functions are called.
Do not worry about which pins the sensors are connected to, that will be handled later.

```ino
/**
 * Returns the acceleration from the Accelerometer in the X direction
 */
int getXAcc()
{
  // Simulate reading from a sensor
  return -1;
}

/**
 * Returns the acceleration from the Accelerometer in the Y direction
 */
int getAccY()
{
  // Simulate reading from a sensor
  return -1;
}

/**
 * Returns the acceleration from Accelerometer in the in the Z direction
 */
int getAccZ()
{
  // Simulate reading from a sensor
  return -1;
}

/**
 * Initalized the accelerometer and returns a 0 value if successfull.
 * @return 0 on success, non-zero otherwise.
 */
int setupAccelerometer()
{
}

```

R11an, 2025-03-15:
[ADXL345 functions.docx](https://github.com/user-attachments/files/19266946/ADXL345.functions.docx)

R11an, 2025-03-15:
@luis-c465  I've completed the functions for the ADXL345 accelerometer, let me know if there is anything i need to fix or add

luis-c465, 2025-03-17:
Great job, all the code looks good.

Is it possible for the code to be submitted as a pull request if you know how?

closed 2025-03-17 by luis-c465