FIU-SEDS/Vensors #32: Generate ideas of how to to detect the start and ending of appogee

luis-c465, 2025-04-01:
Apogee lasts about 5-10 seconds in total.

At the center of apogee the acceleration due to gravity is around 0 m/^2.
The machine needs to be turned on before the start of apogee, preferably 5 seconds before the acceleration due to gravity reaches 0 m/s^2.

Making the program detect the center of apogee is easy, though the device needs to be activated beforehand.

Given the above find a way to detect the payload is around 5 seconds before experiencing apogee, and when the payload is around 5 seconds after experiencing apogee.

Shamar12334, 2025-04-06:
@luis-c465  https://docs.google.com/document/d/1ASBXfW8-nu7BV-wU2Uwbdhk_bzRPFBMDBSj-2M4sBBU/edit?usp=sharing

luis-c465, 2025-04-07:
This is a really good start!

@Shamar12334 

When you get a chance could you expand on to this exactly what new steps would have to be taken in order to detect when appogee is approaching.

For example as the sensors we have only measure acceleration we would first have to measure the integral of acceleration to find the derivative, and so on.

Shamar12334, 2025-04-07:
Okay thanks. I’ll add to it

closed 2025-04-16 by luis-c465