FIU-SEDS/Vensors #33: Write function to filter out outliers from the collected time of flight senosrs

luis-c465, 2025-04-02:
Fill in the bellow code

```ino
#include <vector>

#define TIME_OF_FLIGHT_OUTLIER_CUTOFF 150
#define TIME_OF_FLIGHT_OUTLIER_START 40

/**
 * Filters outliers from list of measurements from the time of flight sensor
 *
 * Filters out values that are either too large or too small
 *
 * Additionally this should implement the z score algorithm to remove outliers that come out
 * This should be fine tuned to keep all data in the range of noise of +- 3mm
 */
std::vector<Measurement> filterOutliersFromTimeOfFlight(std::vector<Measurment> data)
{
}
```

luis-c465, 2025-04-02:
0.00,0.00,2.00,9.00,15.00,21.00,33.00,44.00,55.00,69.00,86.00,105.00,
1.00,1.00,4.00,6.00,13.00,20.00,30.00,43.00,57.00,72.00,86.00,107.00,
1.00,-1.00,4.00,7.00,12.00,22.00,33.00,42.00,56.00,72.00,86.00,105.00,
-1.00,2.00,4.00,9.00,14.00,21.00,32.00,43.00,54.00,71.00,86.00,106.00,
0.00,-1.00,3.00,8.00,13.00,23.00,30.00,43.00,56.00,71.00,86.00,107.00,
0.00,1.00,3.00,6.00,15.00,23.00,30.00,44.00,57.00,71.00,86.00,107.00,
0.00,0.00,4.00,7.00,14.00,23.00,32.00,44.00,56.00,72.00,86.00,106.00,
1.00,2.00,4.00,9.00,12.00,23.00,31.00,43.00,56.00,72.00,86.00,107.00,
-2.00,2.00,2.00,6.00,12.00,20.00,31.00,42.00,56.00,71.00,88.00,105.00,
-2.00,2.00,4.00,6.00,14.00,23.00,33.00,41.00,57.00,71.00,88.00,107.00,
1.00,-1.00,2.00,7.00,13.00,23.00,30.00,41.00,56.00,70.00,86.00,106.00,
1.00,-1.00,4.00,9.00,15.00,20.00,30.00,43.00,54.00,72.00,86.00,105.00,
-2.00,0.00,4.00,7.00,15.00,23.00,32.00,44.00,56.00,69.00,88.00,104.00,
0.00,0.00,4.00,8.00,12.00,20.00,30.00,41.00,57.00,71.00,86.00,104.00,
0.00,1.00,3.00,9.00,13.00,23.00,32.00,44.00,57.00,72.00,86.00,107.00,

Example sensor data which was recorded at every 0.05s where the values are in mm

luis-c465, 2025-04-02:
struct Measurement
{
  /** Distance in mm from the sensor */
  double distance;

  /** Time in Nano seconds since the microcontroller started */
  uint64_t time;
};

JabbarM35, 2025-04-04:
#include <vector>
#include <iostream>
#include <cmath>
#include <cstdint> // for uint64_t

#define TIME_OF_FLIGHT_OUTLIER_CUTOFF 150
#define TIME_OF_FLIGHT_OUTLIER_START 40
#define ZSCORE_MARGIN 3.0 // 3 mm margin around z-score range

struct Measurement
{
    /** Distance in mm from the sensor */
    double distance;

    /** Time in Nano seconds since the microcontroller started */
    uint64_t time;
};

// Function to filter out outliers based on distance and z-score
std::vector<Measurement> filterOutliersFromTimeOfFlight(std::vector<double> rawDistances)
{
    std::vector<Measurement> data;
    const uint64_t intervalNs = 50000000; // 0.05 seconds in nanoseconds

    // Convert rawDistances to Measurement objects with time values
    for (size_t i = 0; i < rawDistances.size(); ++i) {
        Measurement m;
        m.distance = rawDistances[i]; // Use raw distance without noise addition
        m.time = i * intervalNs;
        data.push_back(m);
    }

    std::vector<Measurement> withinThreshold;

    // Step 1: Filter based on absolute thresholds (no margin here)
    for (const auto& m : data)
    {
        if (m.distance >= TIME_OF_FLIGHT_OUTLIER_START && m.distance <= TIME_OF_FLIGHT_OUTLIER_CUTOFF)
        {
            withinThreshold.push_back(m);
        }
    }

    // Step 2: Calculate mean distance
    double sum = 0;
    for (const auto& m : withinThreshold)
    {
        sum += m.distance;
    }

    double mean = withinThreshold.empty() ? 0 : sum / withinThreshold.size();

    // Step 3: Calculate standard deviation
    double variance = 0;
    for (const auto& m : withinThreshold)
    {
        variance += std::pow(m.distance - mean, 2);
    }

    double stdDev = withinThreshold.empty() ? 0 : std::sqrt(variance / withinThreshold.size());

    // Step 4: Calculate the z-score threshold range
    double zScoreLower = mean - 2 * stdDev;
    double zScoreUpper = mean + 2 * stdDev;

    // Apply the margin to the z-score threshold range
    zScoreLower -= ZSCORE_MARGIN;
    zScoreUpper += ZSCORE_MARGIN;

    // Step 5: Filter based on z-score with a margin applied
    std::vector<Measurement> filtered;
    for (const auto& m : withinThreshold)
    {
        if (m.distance >= zScoreLower && m.distance <= zScoreUpper)
        {
            filtered.push_back(m);
        }
    }

    return filtered;
}

int main() {
    // Sample distance readings (in mm)
    std::vector<double> rawDistances = {
        0.00,0.00,2.00,9.00,15.00,21.00,33.00,44.00,55.00,69.00,86.00,105.00,
1.00,1.00,4.00,6.00,13.00,20.00,30.00,43.00,57.00,72.00,86.00,107.00,
1.00,-1.00,4.00,7.00,12.00,22.00,33.00,42.00,56.00,72.00,86.00,105.00,
-1.00,2.00,4.00,9.00,14.00,21.00,32.00,43.00,54.00,71.00,86.00,106.00,
0.00,-1.00,3.00,8.00,13.00,23.00,30.00,43.00,56.00,71.00,86.00,107.00,
0.00,1.00,3.00,6.00,15.00,23.00,30.00,44.00,57.00,71.00,86.00,107.00,
0.00,0.00,4.00,7.00,14.00,23.00,32.00,44.00,56.00,72.00,86.00,106.00,
1.00,2.00,4.00,9.00,12.00,23.00,31.00,43.00,56.00,72.00,86.00,107.00,
-2.00,2.00,2.00,6.00,12.00,20.00,31.00,42.00,56.00,71.00,88.00,105.00,
-2.00,2.00,4.00,6.00,14.00,23.00,33.00,41.00,57.00,71.00,88.00,107.00,
1.00,-1.00,2.00,7.00,13.00,23.00,30.00,41.00,56.00,70.00,86.00,106.00,
1.00,-1.00,4.00,9.00,15.00,20.00,30.00,43.00,54.00,72.00,86.00,105.00,
-2.00,0.00,4.00,7.00,15.00,23.00,32.00,44.00,56.00,69.00,88.00,104.00,
0.00,0.00,4.00,8.00,12.00,20.00,30.00,41.00,57.00,71.00,86.00,104.00,
0.00,1.00,3.00,9.00,13.00,23.00,32.00,44.00,57.00,72.00,86.00,107.00
    };

    // Run the filter
    std::vector<Measurement> filtered = filterOutliersFromTimeOfFlight(rawDistances);

    // Display filtered data
    std::cout << "Filtered Measurements:\n";
    for (const auto& m : filtered) {
        std::cout << "Distance: " << m.distance << " mm, Time: " << m.time << " ns\n";
    }

    return 0;
}

luis-c465, 2025-04-05:
Thanks for sending this over!

I had to make a couple to tweaks to the exact values of things to filter everything out though that was expected.

Great work!

closed 2025-04-05 by luis-c465